dplasa/FTPClientServer #2: Add automated CI or job or script to check build against core

devyte, 2020-05-29:
To allow easy maintenance, and as mitigation to have this lib break, I suggest adding either a CI job or some other mechanism to cover building against the core.

dplasa, 2020-05-29:
Wow, that sounds really cool. Is there a cook-book or a howto or an example, which I could follow?

devyte, 2020-05-29:
I understand there are several ways to do it, and several CI providers possible. The core uses Travis CI. There is also Azure, github actions, etc. I'm not an expert in the area, and I've never done a set up from scratch myself, so I'm not the best person to ask. I think the easiest way is to implement a script that does what you want, then define a CI job that runs it.
You should be able to find tutorials, it's a common enough thing.