dplasa/FTPClientServer #4: Clean up macro pseudo functions

devyte, 2020-05-29:
At a glance I saw some macros used like functions. They should be cleaned up to actually be functions.

dplasa, 2020-05-29:
I used those macros to pass variadic arguments to the printf_P() call. Glad to do it in actual functions but I don't see how. Suggestions?

devyte, 2020-05-29:
Look up variable arguments in C/C++, i. e. `myFunction(int blah, ...)` (literally 3 dots as arg). In particular, the well known printf family is implemented with variable number of args. I think this is what you really intended.

dplasa, 2020-05-29:
Yes sure. But passing variadic arguments from one function (e.g. from ``myFunction(const char* fmt, ...)`` to ``Print::printf(const char* fmt, ...)``) doesn't really work. 

That is, unless there would be a ``Stream::vprintf(const char *fmt, va_list arg)`` function available. But as far as I see, there is none available that supports stdarg.h's va_list.

devyte, 2020-05-29:
I'm not sure about the details here, the last time I looked at propagating va was a long time ago. I guess what is needed is something like Print::vprintf and Print::print_P, and maybe refactor current methods if needed. If that is correct, how about a PR for the core, and then clean up here?

dplasa, 2020-05-30:
Not sure if that is all needed, after seeing the implementation of the Print::printf(...) functions, I've changed my code accordingly.

closed 2020-05-30 by dplasa

devyte, 2020-05-30:
The commit looks good.