dplasa/FTPClientServer #7: Fix timeout calculations

devyte, 2020-05-29:
The code that checks timeouts is flawed and reinvents thw wheel. I strongly suggest replacing with polledTimeout::oneShotMs, which is known to be correct and even covered by unit tests, see core examples.

dplasa, 2020-05-29:
That's a neat suggestion and I've done so! Not really sure if I used polledTimeout::oneShotMs in the very intended way but it works expected and now uses that part of the esp8266 core library.

dplasa, 2020-05-29:
Please do comment on my use of the esp8266::polledTimeout::oneShotMs

devyte, 2020-05-29:
At a glance, and without checking your specific logic, it seems ok. There is one case where you're taking a uint16_t for timeout value, I think it should be uint32_t like everywhere else. Fix that and you cam close this.

dplasa, 2020-05-29:
It would not have made a difference as the waitFor(...) call (waiting for an FTP's server response) was never called with more than 10000ms, but I changed it anyways.

closed 2020-05-29 by dplasa