dplasa/FTPClientServer #8: Make LIST output compatible with FuseFS

dplasa, 2020-05-29:
Fuse FTP-FS (fuse filesystem) -- connection is OK but nothing is visible (neither are directories)
```
mkdir /tmp/xxx
curlftpfs ftp://ftp:ftp@192.168.x.x /tmp/xxx
cd /tmp/xxx
ls -al # empty result
touch x # freeze
```
This empty result is probably due to the format of the LIST output -- the way it was implemented is not understood by FuseFS.
Second problem: FuseFS sends "LIST -a", the "-a" gets treated by the server as filename/dirname and it tries to list "cwd/-a" which fails.

Touch freeze... not sure what happens here.

dplasa, 2020-05-30:
OK, so I updated the code for the LIST command to behave like vsftp's LIST command output would look like. Now, curlftpfs (and other FuseFS clients) show proper directory contents and file dates.

dplasa, 2020-05-30:
However, any file operation still fails. touch'ing, cp'ing... just silently freezes until some timeout. Not fully understood, what is going on here.

When connecting to a vsftpd FTP Server, a ``touch file`` becomes a "SITE CHMOD 644 file" command to the server. I've looked at the debug output of my FTP Server - there is just no command being received. 

Will investigate further.

dplasa, 2020-05-31:
It seems that curlftpfs wants to open a second control connection for file operations like touch, cp, ...
I did not find any command switch to control that behavior and as we support only **1** connection at a time, this does not work out.
Sad to leave it like this but at the time, I don't see how to proceed.
I updated the README accordingly.

d-a-v, 2020-06-01:
Thanks for your investigations and fixes !
About curlftpfs I'll try to see why they try to open a second control connection.

dplasa, 2020-06-01:
I don't know their code structure and maybe it's a dumb idea... but maybe one could try to add an option to limit the number of concurrent connections. There is a "-s single thread" command line option but that doesn't prevent it from using more than one connection.