dplasa/FTPClientServer #17: SD FS unsupported?

theotherjenkutler, 2022-01-07:
Hi Daniel.
Thanks so much for this great library!
I'm working on trying to use it with SD filesystem which appears to work generally the same way code wise as SPIFFS and LittleFS, but it doesn't seem to be able to create a file with the SD filesystem. I'm not sure if this is a bug or just unsupported by the library. I'm using an ESP32WROOM with only 4MB of flash and trying to download a file from FTP that is 5.9MB so there are a lot of things that can go wrong. Not sure the best way to debug other than download starts and then fails with error "Transfer failed after 191 ms, code: 220, descr=220-You are user number 5 of 150 allowed." No file is saved on the SD card. I basically changed all of the references to LittleFS to SD for example: FTPClient ftpClient(SD); bool fsok = SD.begin(csPin); etc. I think that the issue is with SD.begin not being called correctly. Forgive me for asking such a stupid question but how can I view the FTP_DEBUG_MSG information in the serial monitor? I saw another post asking for SD card implementation and I'd love to see if I can figure it out. Thanks again!

theotherjenkutler, 2022-01-08:
This is an issue even using LittleFS or SPIFFS. Enabling verbose debugging and adding #define ESP_DEBUG_PORT Serial at the top of the sketch. For reference, I'm running Arduino 2.0.0 rc-3 (also tried on various other versions of 1.16, 1.18 etc) with a Sparkfun Thing Plus - ESP32WROOM with 16MB Flash and 512k ram. I am attempting to download a file from FTP that is about 2MB. The filesystem works, creates the file and then stops downloading and the file remains with a size of 0 bytes. I can't figure any more out without debugging.  The ESP docs are pretty severely outdated https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html and I'm having a really hard time finding anything about how to address this issue since the DEBUG level is now set in the IDE. I am able to get all the debug logs from the core, WiFiGeneric.h, etc but not from FTP_DEBUG_MSG. Alternately, this can just serve as a bug report that the client example does not work. Thanks!

smnikzad91, 2024-01-14:
Hello everyone, especially the author of this library
I also want to use a SD card for more volume instead of the internal memory. Is there a solution?

smnikzad91, 2024-01-14:
> This is an issue even using LittleFS or SPIFFS. Enabling verbose debugging and adding #define ESP_DEBUG_PORT Serial at the top of the sketch. For reference, I'm running Arduino 2.0.0 rc-3 (also [...]

Hi dear friend
Did you make any progress?