dplasa/FTPClientServer #20: LittleFSSample fails to compile

Andy2051, 2022-03-08:
Title says it all, get exactly the same errors when trying to use FTPClientServer in my own project.

`In file included from /home/andy/Arduino/libraries/FTPClientServer-master/FTPServer.h:32,
                 from /home/andy/Arduino/sketches/LittleFSSample/LittleFSSample.ino:24:
/home/andy/Arduino/libraries/FTPClientServer-master/FTPCommon.h:11:7: error: 'esp8266Pool' has not been declared
   11 | using esp8266Pool::polledTimeout::oneShotMs; // import the type to the local namespace
      |       ^~~~~~~~~~~
In file included from /home/andy/Arduino/libraries/FTPClientServer-master/FTPServer.h:32,
                 from /home/andy/Arduino/sketches/LittleFSSample/LittleFSSample.ino:24:
/home/andy/Arduino/libraries/FTPClientServer-master/FTPCommon.h:130:5: error: 'oneShotMs' does not name a type
  130 |     oneShotMs aTimeout;  // timeout from esp8266 core library
      |     ^~~~~~~~~
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
`

maxint-rd, 2022-03-11:
Same here. Bummer...

Andy2051, 2022-03-11:
Glad to know it's not just me, if you find a working fix please post what you did. I dumped FTPClientServer in favour of SimpleFTPserver, that works with littleFS. Originally I was using ESP8266FtpServer but making the switch SPIFFS to LittleFS and SPIFFS is hardwired into ESP8266FtpServer.

maxint-rd, 2022-03-11:
The fix in pull request https://github.com/dplasa/FTPClientServer/pull/19 made the example compile. All that was needed was to use the changed file (or copy the changes).
After testing various FTP programs I wasn't quite satisfied (see above). Thank you for your suggestion. I'll have a look at it.

In my ESP8266 project I'm using the excellent WiFiManager library. along with code from the very nice ESP8266Webserver/FSBrowser example and was looking for an easy way to copy multiple files at once from the LittleFS file system to back-up all files that I changed using the fsbrower editor.

Andy2051, 2022-03-11:
Can you be explicit about what you did? I read the link and couldn't see the fix. 
I read
> I kept the namespace the same, just changed the include guards to "#pragma once."

But it didn't mean anything for me.

> and was looking for an easy way to copy multiple files at once from the LittleFS file system to back-up all files that I changed using the fsbrower editor.

SimpleFTPserver will not do that for you, you need an FTP client. That is why I was looking at FTPClientServer. For a previous project I wrote one for myself (improving what someone else had done) so I could ftp files from the esp8266 to my NAS. It's in one of my previous posts - not a regular contributor to this forum, so don't know if you can search "user's posts".

maxint-rd, 2022-03-11:
If you look [here](https://github.com/dplasa/FTPClientServer/pull/19/commits/517f8dbc4f4736fcda2586523499e7d43ddc2578) you can see the .h file needs two changes: 
- remove the #ifdef #define at top and the #endif at end
- add the #pragma at top

Thank you for your suggestion to make the ESP have an automatic backup feature by using it as an ftp-client. My own thought was to simply make it a server to allow a client on windows to copy (download) a bunch of files.

This afternoon I tested the SimpleFtpServer example. Some things worked better, but unfortunately I still had similar issues, making me wonder there's something wrong in my setup. I guess further testing is needed (different clients, other hardware, etc)...

maxint-rd, 2022-03-12:
Did some more testing with SimpleFtpServer and found that in the Windows 7 ftp.exe a directory listing would show incomplete lines. After much debugging I saw that the filename pointer was taken directly as c_str() from dir.fileName(), The buffer of that String apparently would no longer be available after the subsequent call to dir.fileTime(). This caused the generateFileLine() call to use invalid data. After fixing that using an intermediate String variable, the Windows FTP explorer successfully was able to copy a whole folder. I still had problems using FileZilla, but at least it's now easier for me to download a copy of all files on the ESP. 

BTW. I post this here because those issues may be related to the ones I encountered when testing the LittleFSSample in this library.

Andy2051, 2022-03-12:
Can't help, six years since I ran win natively, linux user :)

For reference, this is the ftpclient I did:
[https://forum.arduino.cc/t/arduino-esp8266-nodemcu-ftp-client/567765](url)